prampec/IotWebConf #238: can't ping thingname in network

alirezaimi, 2022-01-28:
Hi 
Why i can't ping dns that set in thingname variable ?
When i use MDNS library myself in my code everything is fine but when removed it to use iotwebconf mdns config instead there is no dns name that can i ping in my network .

prampec, 2022-02-22:
In my experience MDNS is not reliable.
There were some tricks, like introducing some sleep hooks, but I'm not sure about those workarounds.

alirezaimi, 2022-05-22:
I use MDNS with wifi library and it's worked perfectly . but in this library this library doesn't work at all !

prampec, 2022-05-22:
We have made some improvements in mDNS in the latest bugfix release. Please confirm, that you are using v3.2.1
https://github.com/prampec/IotWebConf/releases/tag/v3.2.1

alirezaimi, 2022-05-23:
Thanks, sure, i tested it and post the result .