prampec/IotWebConf #244: wifi_init_default: netstack cb reg failed with 12308

societyofrobots, 2022-03-09:
I'm getting an error I don't understand.

I'm in AP mode and connected via phone without issue. I then disconnect via phone, and it correctly changes to wifi mode.

However, it gives an error and the IP incorrectly becomes 255.255.255.255. It never changes to State 4:
`E (59412) wifi_init_default: netstack cb reg failed with 12308`

Looking up the error, it looks like maybe not enough buffer memory?
https://github.com/espressif/esp-idf/blob/cc98ce3c3ae7f5a20bc7bddb79c3b368dbe7066f/components/esp_wifi/src/wifi_default.c#L54-L57
```
    if ((ret = esp_wifi_internal_reg_netstack_buf_cb(esp_netif_netstack_buf_ref, esp_netif_netstack_buf_free)) != ESP_OK) {
        ESP_LOGE(TAG, "netstack cb reg failed with %d", ret);
        return;
    }
```

It's not a problem with my router or custom ESP32 hardware as I don't have this problem with another simpler firmware I'm using.

Below is the log:

```
Disconnected from AP.
State changing from: 2 to 3
[ 47406][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 13 - AP_STADISCONNECTED
[ 47425][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 11 - AP_STOP
[ 47425][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 10 - AP_START
E (59412) wifi_init_default: netstack cb reg failed with 12308
[ 47434][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 11 - AP_STOP
Connecting to [My_Router] (password is hidden)
WiFi timeout (ms): 30000
[ 47473][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 0 - WIFI_READY
[ 47479][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 2 - STA_START
State changed from: 2 to 3
[ 51412][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[ 51414][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[ 51416][D][WiFiGeneric.cpp:880] _eventCallback(): STA IP: 255.255.255.255, MASK: 255.255.255.255, GW: 255.255.255.255
WiFi connected
IP address: 255.255.255.255
State changing from: 3 to 4
Accepting connection
State changed from: 3 to 4
```

Pandananana, 2022-12-19:
I don't know if this is of any help, but I had the exact same issue and fixed it by removing 
WiFi.softAPdisconnect (true);

technicalyarana, 2023-07-01:
01:42:08.740 -> ets Jun  8 2016 00:22:57
01:42:08.740 -> 
01:42:08.740 -> �&
01:42:08.740 -> ����uD��ESP32 Hotspot khula hua hai. Ab aapke phone se connect karein.
01:42:09.612 -> Hotspot SSID: ESP32_Hotspot
01:42:09.612 -> Hotspot Password: password
01:42:28.596 -> Phone connect hua hai.
01:42:28.629 -> Ghar ke Wi-Fi ka SSID aur password enter karein.
01:42:33.632 -> E (31756) wifi_init_default: netstack cb reg failed with 12308
01:42:34.690 -> 

i have giving some issue anyone can solve please

minou65, 2023-08-04:
Was this issue ever resolved?

societyofrobots, 2023-08-07:
I ended up dropping the project for unrelated reasons, so never got back to resolving this issue.

It is not occurring in my other ESP32 projects.

Rob58329, 2024-05-30:
For me this message was caused when a client was connected to the AP, and "WiFi.softAPdisconnect(true)" disconnected it, but the client tries to reconnect but is refused, and so you get the "wifi_init_default: netstack cb reg failed with 12308" message before "WiFi.mode(WIFI_OFF)" completely closes the interface.

If you remove the "WiFi.softAPdisconnect(true)" then the client stays connected until "WiFi.mode(WIFI_OFF)" closes the interface, so you don't get the message.  But I'm not sure if it's better to leave "WiFi.softAPdisconnect(true)" in or to remove it.