prampec/IotWebConf #252: Sequencing mDNS after the device is 'online'

BoraSport, 2022-04-19:
Initializing mDNS needs to happen once the device is online. Once that condition is met, mDSN can bind to WiFi and perform name resolution. This [call](url=https://github.com/prampec/IotWebConf/blob/b89ac9f61b2022cc0c794a74751c5019b4679eef/src/IotWebConf.cpp#L95) needs to be delayed until the device is online. To get this to work in my project (ESP32-S2 using Arduino Framework) I put a catch in the loop to check the state and mDNS setting, if the device is on-line and mDNS is not active it creates the service. 
`// -- Initializing mDNS after WiFi is online`
`if (iotWebConf.getState() == iotwebconf::OnLine && MDNS.queryService("http", "tcp") == 0) {`
`    MDNS.begin(iotWebConf.getThingName());`
`    Serial.println("On-line, initalizing MDNS");`
`    MDNS.addService("http", "tcp", 80);`
`    Serial.printf("Total active MDNS services: %d \n", MDNS.queryService("http", "tcp"));`
`  }`

This runs in `void loop()`. It isn't efficient, but it is reliable.

prampec, 2022-04-20:
This might solves to problem for ESP32, but as far as I see, ESP8266 is a different story.
I'm not very good in mDNS, and it appears to be a more complex problem. E.g. what if we get a different ip address after reconnecting into a lost WiFi network? We might want to end() and then begin() again on every connection.
We must test this. I was hoping for some help in this topic, but probably this is the first one.

BoraSport, 2022-04-20:
The sequencing makes sense. Unfortunately, I don't have an ESP8266 to test with. I don't know your project that well, the state manager looks like a place to call the mDNS setup. Looking through the code [this](url=https://github.com/prampec/IotWebConf/blob/b89ac9f61b2022cc0c794a74751c5019b4679eef/src/IotWebConf.cpp#L734) is where I would try to put the `MDNS.begin(iotWebConf.getThingName());`. I'm not sure where the end operator would need to live.

shmick, 2022-04-21:
The mDNS issue on ESP8266 is that it needs MDNS.update to run in the main loop. I've create a PR for this at #253

prampec, 2022-04-21:
ESP8266 part was merged into the main line and will be part of the next release.
Can you please provide a tested pull request that does the MSDN.begin() when OnLine, and MDNS.end() when offline?!

BoraSport, 2022-04-22:
I'm working on a fix now. I should have it fully tested tomorrow (on ESP32) and I'll submit a pull request.

prampec, 2022-04-28:
Merged into main line. Will be available in version v3.2.1 and higher.

prampec, 2022-04-28:
Ohh, and thank you!

closed 2022-04-28 by prampec