prampec/IotWebConf #257: Saving a parameter results in  ascii characters like '�����'

BoraSport, 2022-05-24:
Using the MQTT Relay example the parameter for the MQTT Server seems to be storing the value incorrectly. 

The parameter shows up and says the value is stored but this is what comes back in the serial monitor.
|   \-- 'mqttServer' with value: '���������������������������������������������������������������������������������������������������������������������������������'

I'm on an ESP32-S2 (Sparkfun Thing Plus WRL-17743). All the normal configuration parameters work great. This is all using the example [here.](https://github.com/prampec/IotWebConf/blob/master/examples/IotWebConf07MqttRelay/IotWebConf07MqttRelay.ino)

The /config page does initially show the MQTT Server value correctly. After a refreshing the home page, the MQTT Server value updates to "яяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяЂ"

It is interesting that both values are 129 characters.  For testing purposes the value I'm inputting for this parameter is "test.mosquitto.org".

prampec, 2022-05-25:
Cannot confirm this on my ESP8266 (D1 Mini). I see the following log after startup:
```
Starting up...
Config version: mqt1
Config size: 549
Loading configurations
[iwcAll]
|-- [iwcSys]
|   |-- 'iwcThingName' with value: 'testThing'
|   |-- 'iwcApPassword' with value: <hidden>
|   |-- [iwcWifi0]
|   |   |-- 'iwcWifiSsid' with value: 'ede'
|   |   \-- 'iwcWifiPassword' with value: <hidden>
|   \-- 'iwcApTimeout' with value: '30'
|-- [iwcCustom]
|   \-- [mqtt]
|       |-- 'mqttServer' with value: 'test.mosquitto.org'
|       |-- 'mqttUser' with value: 'asd'
|       \-- 'mqttPass' with value: <hidden>
\-- [hidden]
Ready.
```

BoraSport, 2022-05-25:
@prampec thanks for your investigation. I went back through and pulled a clean set down from GitHub and retested. This was a typo on my side in a block I didn't realize I had changed. My apologies. Thanks again.

closed 2022-05-25 by BoraSport