prampec/IotWebConf #259: Question: Load/save config json from spiffs and assign params to local variables

simplest-circuits, 2022-06-12:
Hello,
my typical application should work like this:
- check jumper pin  and existence of json in spiffs -> if true, load json config and assign the config params to global variables in rest of my code
-> else: create WiFi AP and load iotwebconf, after save json to spiffs

The WiFi is only needed for configuration and if jumper is set, in any other case WiFi needs to be disabled (power saving)

How can load the json and assign the values from the json to my local/global variables?
The custom params section in the iotwebconf is already implemented in my code only the assignment of the  dealing with json config is a big question mark for me.

Thanks to all for this great project.

prampec, 2022-06-12:
For the JSON please check out example IotWebConf17JsonConfig.
Assignment can be done by a simply strncpy + saveConfig()

simplest-circuits, 2022-06-12:
Thanks for the answer, but where are the loaded config params saved after loading it from the json to use the strncpy command?
And if I understand the code correctly, than each time after the json config is loaded, it will be also saved in the eeprom, why?
For simplicity it would be easier to load the json, deserialize it and save each params to a (global) struct or a custom class: configvalue1, configvalue2,configvalue3....

prampec, 2022-06-13:
"load json" means: load configuration from file and store in eeprom
iotWebConf does two main functionalities: 1. provide portal and store configuration parameters; 2. manage WiFi connection state.
You must define your own custom parameters aswell there are defines system parameters, that are handled under functionality#1. So for the question "where are the loaded config params saved?" the answer is: in the system parameter instances and in your custom parameter instances. Each of these utilized with a static string storage for storing string values (valueBuffer).
You might want to check the Hacking guide and especially section 'Accessing system properties':
https://github.com/prampec/IotWebConf/blob/master/doc/HackingGuide.md#accessing-system-properties

simplest-circuits, 2022-06-14:
I understand the general functionality, but it´s super confusing for me to handle the customParams i´ve already setup in my own webconfig.h to extend the systemParams with e.g. params for Telegram (TELEGRAMBOTtokenParam, TELEGRAMChatIDParam, etc..)
The HTML/WebGui has all my extended params included, that was no big problem, with the example "IotWebConf03CustomParameters".
Now I have to deal with these params (mycustomclass.getTelegramBotTokenValue()) via methods or direct access on the values of the params for an example...
Option 1: Load the params directly from the json config (which iotwebconf method handle the transfer between eeprom and assignment to a variable? -> probably not recommended, because it will automatically load into the eeprom..?
Option 2: Load directly from eeprom (in which address will the params be saved, because it´s automatically done) -> more informations required for me to be able to deal with the eeprom (because it´s "invisible" and i don´t know how the data is saved there)

Sorry for my stupid questions, but a little hard to understand your thoughts and the way of walk in the code.
Probably you can share another example how the workflow can look like. please?