prampec/IotWebConf #266: Won

AardvarkMan, 2022-08-21:


BrainHunter, 2022-10-16:
Hi there, 

I have the same problem with a lolin s2 mini board. 
initial AP works fine but then connecting to my wifi is not possible.
what i also see is: when the board switches back to AP after the timeout it is not visible to any device. 
seems that reconfiguring from initial Ap to STA back to AP breaks wifi.

also tried some arduino wifi examples: No problem connecting to my wifi there.

BrainHunter, 2022-10-16:
> I have the same problem with a lolin s2 mini board. initial AP works fine but then connecting to my wifi is not possible. what i also see is: when the board switches back to AP after the timeout it is not visible to any device. seems that reconfiguring from initial Ap to STA back to AP breaks wifi.

update from my side: 
i just updated my esp32 platform package (now V5.2.0) within platformio and now it seems to work!

AardvarkMan, 2023-03-16:
I built the code in PlatformIO with the 6.1.0 platform package and it is working great. I guess it is something in the Arduino compile process.

I'm not sure how to fix this for the Arduino IDE, but PlatformIO is a working solution.

closed 2023-03-16 by AardvarkMan