prampec/IotWebConf #276: Unable to commit messages on EEPROM with this library

jgarcia-atg, 2023-05-25:
Hello.

I am using IotWebConf library for configuring Wi-Fi and MQTT connection for a device based on ESP8266.
Unfortunatelly, I have to write myself some parameters on EEPROM but when y commit the write, it fails. For example:

>  EEPROM.write(MemWifi,48);    // (default: 5;  maximum: 10)
>  EEPROM.write(MemMQTT,48);    /(default: 5;  maximum: 10)
>  EEPROM.write(MemADC,48);    //(default: 5;  maximum: 10)
>  EEPROM.write(MemAHT,48);    //(default: 5;  maximum: 10)
>     if (EEPROM.commit()) {
>        Serial.println("EEPROM successfully committed");
>   } else {
>        Serial.println("ERROR! EEPROM commit failed");
>    }

MemWiFi and so on are previously defined like:
 > #define MemWifi 711

The problem is that when I call EEPROM.commit() it returns false, so I read on serial terminal "ERROR! EEPROM commit failed"
I saw another Issue (https://github.com/prampec/IotWebConf/issues/60) and I tried his solution, but I'm still unable to commit the changes.

Any thoughs about this?

Thank you.

prampec, 2023-05-25:
There was a fix, that fixed EEPROM was not closed after startup: https://github.com/prampec/IotWebConf/commit/b89ac9f61b2022cc0c794a74751c5019b4679eef
Maybe that introduced new problem. E.g. it might happened, that EEPROM.end() was called, that disabled any further eeprom writes?

jgarcia-atg, 2023-05-31:
Hi @prampec .

Thanks for your reply.
Do you mean that maybe I have to modify .cpp file?

prampec, 2023-05-31:
Do you properly use EEPROM.begin() and EEPROM.end() ?