prampec/IotWebConf #280: Is a description of parameters available ?

Greece2, 2023-10-12:
Hi,
I tried to find the answers in the examples, the documentation, the source code etc., without success. Possibly because
everyone thinks this is obvious and self-explanatory ;-)

Is there a description of the following parameters ?
`IotWebConfTextParameter stringParam = IotWebConfTextParameter("String param", "stringParam", stringParamValue, STRING_LEN);`
Starting from the left:
- stringParam holds the entered string ?
- "String param" seems to be the text that is displayed in the form
- "stringParam" does what ?
- stringParamValue does what ?
- STRING_LEN is the max. number of characters

is it possible to have a default string?  And if so, where and how would that be placed ?

Same question for this 
`IotWebConfNumberParameter intParam = IotWebConfNumberParameter("Int param", "intParam", intParamValue, NUMBER_LEN, "20", "1..100", "min='1' max='100' step='1'");`
- intParam holds the entered integer value ?
- "Int param" seems to be the text displayed in the form 
- "intParam" does what ?
- intParamValue does what ?
- NUMBER_LEN seems to be the max. number of digits
- "20" seems to be the default value
- "1..100" seems to be a text displayed in the form
- "min='1' max='100' step='1'" seems to set the possible range

Same questions for all other possible enties like:
```
IotWebConfParameterGroup group1 = IotWebConfParameterGroup("group1", "");
IotWebConfNumberParameter floatParam = IotWebConfNumberParameter("Float param", "floatParam", floatParamValue, NUMBER_LEN,  nullptr, "e.g. 23.4", "step='0.1'");
IotWebConfCheckboxParameter checkboxParam = IotWebConfCheckboxParameter("Check param", "checkParam", checkboxParamValue, STRING_LEN,  true);
IotWebConfSelectParameter chooserParam = IotWebConfSelectParameter("Choose param", "chooseParam", chooserParamValue, STRING_LEN, (char*)chooserValues, (char*)chooserNames, sizeof(chooserValues) / STRING_LEN, STRING_LEN);
```

As one usually has more than one parameter of each type (like 2 integers for upper and lower limits, several strings for name, street, address etc.), how would one do that ?
As all the examples always only show one of each type, it would be nice to have an example showing how to have more than one integer etc.

I know, it seems to be obvious to everyone else but me ;-)  Still, I'd like to make some progress and continue to learn :-)

Thanks and best regards !

marcelstoer, 2023-10-29:
Maybe you're looking for the code documentation in https://github.com/prampec/IotWebConf/blob/master/src/IotWebConfParameter.h?

- for the `Parameter` super class: https://github.com/prampec/IotWebConf/blob/master/src/IotWebConfParameter.h#L181-L192
- for e.g. the `TextParameter` sub class: https://github.com/prampec/IotWebConf/blob/master/src/IotWebConfParameter.h#L229-L236

Greece2, 2023-11-09:
Thanks, that's something to start with !
But having to find comments in several hundred (or even thousands) lines of source code is not that easy and no "real" documentation.
Also, in the examples, having the same words (like "intparam") used in several parts of the parameters is a bit confusing when they have adifferent meaning / purpose.